ShikyC/Taiko-Drum-Controller #12: Update 2 drum mode for other boards

ShikyC, 2024-01-05:


sorrowfullys, 2024-01-27:
新更新的pcb文件已经下单制作了，元器件也买好了，新入手了esp32 ，想问下这个电路板做好后，也可以刷单人键盘模拟么，[ESP32-S3.ino]这个是否适用您新增的pcb

ShikyC, 2024-01-28:
@sorrowfullys 哇你这速度也太快了，我也才刚在嘉立创下单……理论上依然可以模拟键盘，你把不过在这之前有个很重要的事情是你要的ESP32-S3开发板的尺寸和针脚和我的要一样才行（22针脚长，10针脚宽，以及针脚的顺序也要一致。我这块PCB不适配ESP32和ESP32-S2，只能用S3的。

ShikyC, 2024-01-28:
@sorrowfullys 我用的是这块板子：[image]

sorrowfullys, 2024-01-28:
太好了，中文交流，应该是一样的，单排也买的22p的，可调电阻我买的10k和50k的，我自己设计的pcb没有可调电阻，用的32u4，发现改灵敏值后，键位就乱窜，哪怕改0.1都不行，不知道什么原因，然后那个可视化观察数值的，一直反复快速刷新，观察不到有效数字，可以探讨交流下么，qq541784531，街机鼓的木板尺寸我已经还原了，也定制了，我做的是带钢板的版本，包括减震垫的非标规格也有厂家可以做，直型的减震垫太硬了，虽然便宜
[image]

ShikyC, 2024-01-29:
@sorrowfullys 我重构整合了一下，之前数值乱跳以及有一侧没反应应该是因为有个针脚被我改来改去搞乱了，现在应该好了……过两天会抽空更新一下文档详细说一下怎么用（你也可以看注释自己调试一下先）。
时间关系我现在只测试了两个模式（单鼓实时调节灵敏度/双鼓预烧录灵敏度）在虹CHN版+原生力度感应输入（替换我提供的那个dll文件）基本完美了，但是模拟成键盘玩次郎或者模拟器之类的我还没来得及测，不确定还能不能用

sorrowfullys, 2024-01-29:
@ShikyC 新的你设计的那个电路板没有问题对吧，照做照焊就行

ShikyC, 2024-01-30:
@sorrowfullys 理论上没问题，不过我还没收到打样，不敢100%说没问题……

sorrowfullys, 2024-01-30:
@ShikyC 你有没有看那个taiko input的项目，感觉那个方案很完善，只是电路板焊接可能需要加热平台
你这个板子应该明天到，其他配件也这两天到，到时候刷写调试代码可以咨询下你么

sorrowfullys, 2024-01-30:
@ShikyC  qq  541784531