ShikyC/Taiko-Drum-Controller #16: 敲擊感應電路相關問題

HellbroWDR, 2024-02-27:
作者您好，我參考了您的文檔搭建了使用橋式整流器的電路。使用的板子是arduino leonardo，橋式整流器使用的是DB105(店家DB107缺貨)，電阻使用100k歐姆。
目前遇到的問題是，沒有裝橋式整流器的話，敲擊可以感測到，但是有裝的話就感應不到敲擊。想請問作者有遇到這種狀況嗎？

ShikyC, 2024-02-29:
Please first check the output of the rectifier (with a multimeter or an oscilloscope) to verify it's working. Also, please note although the input polarity doesn't matter, the output DOES. You should connect `+` to an analog input pin and `-` to the ground. Can you show me a picture of your circuit?

HellbroWDR, 2024-02-29:
[image]
棕線接analog input，黑線接GND

ShikyC, 2024-03-01:
Can you test the direct output of the rectifier with a multimeter? Using voltage mode, measure the value between `+` and `-` while you hit the drum. Although not accurate, you should be able to see a value change. If not, the rectifier chip might not work for you.
You can also remove the resistors and connect `+` directly to the analog input of your Leonardo board (unlike ESP32, ATmega32 boards measure analog inputs from 0 to 5V so you don't need to scale down the input voltage) to ensure that it wasn't the issue of the resistors.