ShikyC/Taiko-Drum-Controller #18: Getting constant output from Serial Monitor in DEBUG mode

mechachoi, 2024-04-03:
Hello ShikyC,

Thank you so much for developing this. I am very passionate about getting my own taiko setup up and running. 

I have my Taiko (an official arcade taiko) connected to my circuit with bridge rectifiers. I am seeing approximately 4V constant voltage out of the output of the rectifiers when I put my multimeter across it, but I do see a change when I strike the drum in the respective locations.

However, whenever I open the serial monitor to see what is happening in DEBUG mode, I am getting basically constant values from the Serial Monitor, 126945 just over and over again. I'm trying to figure out what's going on that would cause this behavior. 

If I change one of the sensitivities of one of the sections of the drum, all the values change.

mechachoi, 2024-04-04:
Doesn't the piezo drum need some kind of powered circuit to output signal?

ShikyC, 2024-04-04:
No. Unlike active input sources (such as microphones), piezoelectric sensors produces periodic voltage by themselves as if they're power sources.