ShikyC/Taiko-Drum-Controller #19: Request for Centroid (Pick-and-Place) File

Dennisjoch, 2024-05-17:
Hello,

I am currently using the PCB design files from this project, but I need a Centroid (Pick-and-Place) file for assembly purposes. Unfortunately, I only have the Gerber files and do not have the original PCB design files to generate the Centroid file myself.

Would it be possible for you to provide a Centroid file for this project?

Thank you very much for your assistance!

Best regards, Dennis Joch

ShikyC, 2024-05-22:
Unfortunately I have been modifying the origianl PCB project recently for a new release so I can't generate the Centroid fild for this version.

My suggestion is, because the PCB is so simple that you can buy the components listed in the BOM file and hand-solder it. Or if you need it urgently and live in the U.S., I can ship one to you for free (excluding the shipping fee).

Dennisjoch, 2024-05-23:
Hey, it's all good. After some head-scratching and overpriced offers on Fiverr, the customer support from PCBWay got in touch and was able to help me based on your project preview image and my labeling. Just as I was about to cancel the assembly because I thought I'd have to solder it myself. Thank you for your willingness to help, and great project! Thanks.

closed 2024-05-23 by Dennisjoch