ShikyC/Taiko-Drum-Controller #22: 关于传感器信号干扰到其他传感器的问题

Steve-xmh, 2024-07-15:
大佬你好，首先非常感谢大佬能够提供这样的一个项目代码供参考！我参考了你的源代码编写了基于 ESP-IDF 框架的模拟信号实现，但是遇到了传感器信号（串音）相互干扰的问题（敲击某个传感器时，会影响其他传感器的数值，即便其他传感器并没有被敲击），想询问大佬可能的问题和错误的检查流程。

我使用的传感器是 muRata 的震动传感器（应该是官方街机使用的版本，为了方便连接我改装并延长了原先的线材，引成了两根杜邦公头线）：
[image]

然后通过 [Taiko Controller Test](https://shiky.me/taiko) 页面检测的数据大概是这样的（只对右侧鼓面（右咚）的传感器敲击）：

[image]

并且信号在敲击之后是缓慢下降才归零的，不清楚是不是传感器被我改动过了的缘故：

[image]

开发板我使用的是微雪电子的 ESP32-S3-Zero，并将 GPIO1-4 引脚用于连接传感器（正极连接了 3V3 引脚）输入模拟信号。

大概实物连接的样子是这样的：

[image]

如果还需要其他信息请不吝指教，非常感谢！

Steve-xmh, 2024-07-17:
更新一下，我把代码重构成了使用 ADC 持续读取方式，再修改了一下连接方式后不会那么容易出现到干扰其他传感器信号的情况了，但是信号缓慢下降的情况还没有头绪：

（蓝色圈起部分为用手指敲击传感器的时间刻）
[image]

ShikyC, 2024-07-20:
不好意思来晚了，最近刚毕业入职有点忙……我不确定这种信号缓慢damping的现象是不是振动传感器上其他元器件导致的（我的板子设计没考虑这些因素，就是简单的让压电陶瓷片直连ADC），但如果这是官方鼓用的传感器的话那可能从压电陶瓷出来会有匹配阻抗或者滤波的元件。你能发一下这个传感器的data sheet吗，我这周末有时间看一下

Steve-xmh, 2024-07-20:
> 不好意思来晚了，最近刚毕业入职有点忙……我不确定这种信号缓慢damping的现象是不是振动传感器上其他元器件导致的（我的板子设计没考虑这些因素，就是简单的让压电陶瓷片直连ADC），但如果这是官方鼓用的传感器的话那可能从压电陶瓷出来会有匹配阻抗或者滤波的元件。你能发一下这个传感器的data sheet吗，我这周末有时间看一下

感谢大佬回复！我确实是为了想使用接近的材料复现街机的感觉才参考网上的说明买了这样的传感器，因为也是网络上找到看上去类似的就买了所以不清楚具体的型号，不过好像都叫它 muRata PKS Shock Sensor，店家也没有写详细的信息，我购买的[淘宝页面是这个](https://item.taobao.com/item.htm?spm=a1z09.2.0.0.478b2e8dJ8OkCT&id=695496541914&_u=k3oh5kvocede)。

至于 Data Sheet 的话只能尽量找应该是同款的给大佬看看能不能作为参考了： https://octopart.com/datasheet/pks1-4a10-murata-5005343

（其实我是第一次尝试这样的硬件开发所以有很多不太懂的地方，如果哪里写的不是很对还请多多指教）

> 其实昨天刚想起来自己参考代码的时候没有理解那个 ADC 取值的用法，我是直接把 ADC 的取值简单限制范围然后直接当作摇杆输入了，没留意大佬其实是把差值拿来用，等回头电脑在身边的时候我再修改一下代码看看能不能有所改善这个情况。

ShikyC, 2024-07-20:
啊我好像发现你电路的问题了……看图虽然不太好猜，不过听你描述你是不是把3.3V电压加到压电陶瓷片上了？压电陶瓷片本你可以理解为是一个电压源，不需要外接电源的。直接一头接GPIO另一头接地就行了。
考虑到ESP32的ADC最大输入只有3.3V，压电陶瓷的瞬时电压可能会超过这个数（我自己测大概是0～5V的范围）所以可以接两个电阻分压（不用也行，烧毁的风险很低）。
希望能帮到你！我也打算买几个你说的这个传感器回头看看怎么辉石

Steve-xmh, 2024-07-21:
> 啊我好像发现你电路的问题了……看图虽然不太好猜，不过听你描述你是不是把3.3V电压加到压电陶瓷片上了？压电陶瓷片本你可以理解为是一个电压源，不需要外接电源的。直接一头接GPIO另一头接地就行了。 考虑到ESP32的ADC最大输入只有3.3V，压电陶瓷的瞬时电压可能会超过这个数（我自己测大概是0～5V的范围）所以可以接两个电阻分压（不用也行，烧毁的风险很低）。 希望能帮到你！我也打算买几个你说的这个传感器回头看看怎么辉石

一语道破，我确实是接到了 3V3 的孔，重新接地之后现在信号下降的情况有了很大的改善。但是敲击一个传感器导致其他传感器也一起感应出信号的情况还是存在，还是要找一下原因（图中敲击了鼓边传感器但是鼓面传感器也发生的感应）：

[image]

ShikyC, 2024-07-22:
这个其实不是问题，你只需要通过调整灵敏度，确保每次敲击时，**数值最大的面就是你敲级的面**就行了，游戏内置的IO机制会自动触发这个敲击并忽略其他敲击。啊当然前提是你玩的是你懂的那个08.18游戏版本并且替换`bnusio.dll`文件。（按理说应该对几个月前新泄露的24版游戏也能用，我还没试过不确定）。
如果你玩第三方游戏（TJAPlayer或者次郎）就要刷模拟成键盘的那个版本的固件了（调试比较麻烦，只能用串口通信），不过逻辑是一样的，保证敲哪边哪个数值最大即可。

Steve-xmh, 2024-07-22:
> 这个其实不是问题，你只需要通过调整灵敏度，确保每次敲击时，**数值最大的面就是你敲级的面**就行了，游戏内置的IO机制会自动触发这个敲击并忽略其他敲击。啊当然前提是你玩的是你懂的那个08.18游戏版本并且替换`bnusio.dll`文件。（按理说应该对几个月前新泄露的24版游戏也能用，我还没试过不确定）。 如果你玩第三方游戏（TJAPlayer或者次郎）就要刷模拟成键盘的那个版本的固件了（调试比较麻烦，只能用串口通信），不过逻辑是一样的，保证敲哪边哪个数值最大即可。

这样子，那我就多花心思去调试一下就行，那议题就差不多就解决了，非常感谢！

closed 2024-07-22 by Steve-xmh

Steve-xmh, 2024-07-22:
啊还是返回来说明一下这个现象，经常出现以下情况：
- 敲鼓边数次，有点反应，然后另外三个传感器一起有感应而且比这个鼓边还高
- 敲鼓面，四个传感器一起有大小不一的感应抖动
- 快速敲击多次，漏拍现象严重
诸如以上的情况，感觉不像是简单调节传感器灵敏度就能解决的情况（）